CTCaer/hekate #780: No main boot entries found

HeJerry, 2022-09-15:
use 5.8 lanch 5.7.2  
launch have issue “
No main boot entries found...
check that bootloader/hekat_ipl.ini has boot entries 
or use More configs button for more boot entries”
but enter More config have issue
"No .ini or boot entries found...
Check that a .ini file exists in bootloader /ini/
and that it contains at least one entry "
i don't know how to fix it

impeeza, 2022-09-15:
Hi, this space is not for support, is for report bugs and issues.

for support please go to proper channels like GBATemp, Discord or similar.

you may have a corrupted/fault SD Card. on the Noob Paradise Thread of GBATemp you will get help easy.

CTCaer, 2022-10-07:
You are expected to make one or put a premade .ini.
None is shipped by default.

If you want to do it manually, you can check several template entries here:
https://github.com/CTCaer/hekate/blob/master/res/hekate_ipl_template.ini

closed 2022-10-07 by CTCaer