CTCaer/hekate #783: Freeze after update to 5.9.0

Heib386, 2022-10-12:
Hekate freezes after update to 5.9.0
Does not react to any input

LeddaZ, 2022-10-12:
Same here, booting 5.9.0 payload with 5.8.0 bootloader folder works

CTCaer, 2022-10-12:
What's bpmpclock in nyx.ini?
If it's 1, try with 2 and report back.

muxi1, 2022-10-12:
I can also confirm this. I use an Erista v1 and the problem occurs irregularly after a reboot from Tegra Explorer or also when booting via a payload transmitter. There is no nyx.ini present in my setting.

LeddaZ, 2022-10-12:
> What's bpmpclock in nyx.ini? If it's 1, try with 2 and report back.

I have 2 already, putting 1 has the same issue.

LeddaZ, 2022-10-12:
[https://i.imgur.com/nHerWsg.png](Image)

arch-box, 2022-10-12:
Erista. 
CPU Speedo 0: 1999
CPU Speedo 1: 3397
CPU Speedo 2: 2005
SoC Speedo 0: 1896
SoC Speedo 1: 0x7F
SoC Speedo 2: 0

CTCaer, 2022-10-12:
possible fix:
[nyx.zip](https://github.com/CTCaer/hekate/files/9768474/nyx.zip)

Whovian9369, 2022-10-12:
Note: Started writing this before the possible fix was posted, so this is meant as a reply to the main issue.

Also having this issue with 5.9.0 and mainly (but not *always*) when I reboot into Hekate from another payload 
For example, I had a freeze when rebooting from Lockpick_RCM 1.9.8 using the "Reboot to Hekate" menu option, and Atmosphere 1.4.0 (pre-release) using the automatic reboot after updating the HOS firmware.
Time, temperature, and batty level all showed as `00:00`/`00.0`/etc when this happened.

Curiously for me, I proceeded to try rebooting to Hekate via LP_RCM and it worked again just fine. 

I'll try and test the possible fix, thanks!

CTCaer, 2022-10-12:
Doubt it matters from where it's injected/loaded.
Hopefully it's what I think it is and is fixed by that bin above.

Whovian9369, 2022-10-12:
> Doubt it matters from where it's injected/loaded. Hopefully it's what I think it is and is fixed by that bin above.
No immediate freeze when rebooting from AMS 1.4.0 (PR) so I'm considering it fixed for myself, thanks!

CTCaer, 2022-10-12:
Thanks for reporting that. I will wait for all of them here and then push it in release.

LeddaZ, 2022-10-12:
The fix works for me too

Heib386, 2022-10-12:
Fix works for me too, thanks

CTCaer, 2022-10-12:
@Glitch-Code-Design @muxi1 @arch-box ?

CTCaer, 2022-10-12:
Pushed fix to release. Anyone having that issue can redownload the release.

closed 2022-10-12 by CTCaer