CTCaer/hekate #784: My switch screen is completely yellow

Rikkamortus, 2022-10-12:
I updated to hekate v5.9.0. I dragged & dropped the bootloader folder into the root of the SD card and I was prompted to replace the bootloader folder that was already there and I replaced it. 

When I put my switch in RCM mode, it accepted the hekate_ctcaer_5.9.0.bin payload and launched the screen but when I try to enter the home screen, the screen becomes all yellow. I don't know what is happening.

CTCaer, 2022-10-13:
I don't get what you mean by "launched the screen" and "enter home screen"
I'm assuming HOS and 15.0.0, which means you also need to update atmosphere.
And make sure you don't have incompatible sysmodules.

Rikkamortus, 2022-10-13:
When I enter the payload in tegra, it will open a menu ([Launch, More Configs, Payloads, EmuMMC](https://switch.homebrew.guide/_images/Step5.png)). Normally (before this most recent update), if I click on Launch, it would bring me to the switch home screen. But now when I do that, the entire screen turns yellow and I have to turn off the switch. 

I don't know what HOS is. 
I thought hekate and atmosphere are the same thing. 
How do I find incompatible sysmodules and what do I do with them when I find them?

CTCaer, 2022-10-13:
So as I thought then.
You forgot to update atmosphere also (drag n drop and replace like how hekate update is).

If you still get that, rename atmosphere folder, paste anew and see if it boots.
If it does, then you have incompatible sysmodules in content folder. Just restore them one by one until you find which one causes that. Then search for a new version of it.

Other than that, here's the wrong place to seek support for that.

closed 2022-10-13 by CTCaer

Rikkamortus, 2022-10-15:
Where is the atmosphere update?

Masamune3210, 2022-10-15:
on their github