CTCaer/hekate #786: hekate_ctcaer_5.9.0.bin as Paylod not reliable w/ Spacecraft-NX

isermann, 2022-10-16:
Hello all,

I discovered that using the new hekate_ctcaer_5.9.0.bin as payload.bin for Spacecraft-NX (SX Core) on patched Erista with HOS 15.0.0 / Atmosphère 1.4.0 on emuMMC is very unreliable. Most of the time booting to OFW or black screen.

Using old hekate_ctcaer_5.**8**.0.bin as payload will load new hekate 5.9.0 & CFW (Atmosphère 1.4.0) reliable again every boot.

Cheers

urherenow, 2022-10-16:
Is your chip flashable? Try the latest hwfly-nx firmware, which was forked from spacecraft-nx.

CTCaer, 2022-10-17:
- How you boot (your hekate_ipl.ini entry or anything else)? Because it's imposible to boot into OFW like that.
- Also describe fully what you see on screen when it works and when it's not.
- Paste a photo or screenshot of microsd info from hekate
- Retry again by redownloading the release, pasting payload.bin and then safely ejecting, in case of previous sd corruption.

Also, hekate v5.8.0 can chainload hekate v5.9.0 as you said, which points to a spacecraft fault. Not the other way around.
Or the payload ended up corrupted when you placed it the first time. 

Copying and setting a payload was "standarlized" in 2018.
Till now I saw many injectors breaking it and corrupting the payloads (and then releasing fw updates to fix that).
And if you corrupt them, they will hang, because of the compressed format some use, like hekate.

Other than that, if you want to find out what's the issue, you can answer the above questions, so I can learn if you even reach hekate or Nyx, when it happens, etc. Or you can pm me in discord (retronx), since I need lots of info from you.

isermann, 2022-10-17:
> * How you boot (your hekate_ipl.ini entry or anything else)? Because it's imposible to boot into OFW like that.
>
>     * Also describe fully what you see on screen when it works and when it's [...]

Thanks for your detailed reply!

I didn't boot into OFW intentionally, maybe the failed Spacecraft-NX payload injection booted CFW on sysNAND, but I did not verify as I didn't want to boot anything else than CFW on emuMMC.

When it didn't boot into CFW, the Switch either showed a black screen or NINTENDO logo followed by booting into OFW w/o any Atmosphère logo.
~30% of the time it loaded hekate (w/ new boot animation) successfully with hekate 5.9.0 as payload.

I did replace hekate several times as payload.bin though. Also re-downloaded from this repo and also tried older versions (which worked). And I also let Windows check for SDcard faults every time before I safely ejected the card.

Meanwhile, as @urherenow suggested, I tried switching from Spacecraft-NX v.0.2.0 to hwfly-nx v.0.7.1 firmware for my original SX Core.
Unfortunately, hwfly-nx 0.7.1 firmware boots only into "toolbox mode (green pulsing LED)" and does only and always show a black screen. Doesn't seem to be compatible with original SX Core on Erista?
But I found a modified [Spacecraft-NX-Mod v3](https://github.com/KazushiMe/Spacecraft-NX-Mod/) which actually does boot hekate 5.9.0 as payload reliably!

For me now, it looks like hekate 5.9.0 introduced a change which Spacecraft-NX v.0.2.0 couldn't handle anymore.

Do you want to continue investigating?

urherenow, 2022-10-17:
On hwfly, I don’t think it would install itself if it wasn’t compatible. Did you use the toolbox to reset and retrain the chip? The timing may be different, because it does boot faster.But since you found a firmware that works, I guess it’s best not to mess with it.Sent from my iPhone

isermann, 2022-10-17:
> On hwfly, I don’t think it would install itself if it wasn’t compatible. Did you use the toolbox to reset and retrain the chip? The timing may be different, because it does boot faster.But since you [...]

I couldn't flash the hwfly-nx firmware via the hwfly-toolbox as I couldn't keep the SX Core awake post boot w/ Spacecraft-NX firmware to enable flashing from within the hwfly-toolbox.
Hence, I flashed hwfly-nx bootloader & firmware via USB directly. After that, the SX Core started re-training and after success it automatically & directly entered "toolbox mode" w/o booting hekate or even turning the screen on while no SDcard. Just a slowly pulsing green LED...

I tried using the serial interface to access the USB tty input mode (blue pulsing LED), which didn't response back though the connection seemed established.

In the end, I couldn't get hwfly-nx firmware booting and replaced it with the modded Spacecraft-NX-Mod v3 which then worked with hekate 5.9.0.

CTCaer, 2022-10-18:
Yeah that behavior is online corruption of the payload and it's the fault of chip/fw/loader.
Find me in discord for a possible testing/mitigation of it.

shigutso, 2022-10-21:
Hi, I am having a similar issue, Hekate 5.9.0 won't boot on my RCM Loader.

This is the device info (got from Linux `dmesg` command):

```
New USB device found, idVendor=14cd, idProduct=0103, bcdDevice= 1.00
New USB device strings: Mfr=0, Product=1, SerialNumber=2
Product: RCM loader
SerialNumber: RCMLoader One V1.2
RCM Loader One plus  2.03 PQ: 0 ANSI: 0 CCS
```

The device came bundled with Hekate 5.6.5 and that works fine. If I update the `payload.bin` file to Hekate 5.9.0 I only get a black screen. I should say that the 5.6.5 payload turns into 5.9.0 because of the newer `bootloader/update.bin` file in my sdcard.
I tried Hekate 5.8.0 and that seems to also work fine. Something in 5.9.0 is preventing my RCM Loader to inject the payload properly. I could not find any firmware updates newer than the one I have for this device (seems to be version 1.2/2.03?).

Let me know if more info is needed! Thanks

impeeza, 2022-10-21:
If your dongle is a rcm loader 2(white cardboard box) you need to upgrade the loader firmware https://www.xkit.xyz/support/

shigutso, 2022-10-21:
It seems my device came with the latest version. I bought it from Aliexpress a couple weeks ago

impeeza, 2022-10-21:
So is a "v5" clone don't need the upgrade

CTCaer, 2022-10-21:
As I said they corrupt the payload. And I bet the issue they have has to do with size alignments.
And I bet they have no clue how to fix it (they already tried once in fw 2.02).

But if no one contacts me in discord to test an actual mitigation, I will just close this issue as **Broken 3rd party implementation**.
And this will keep coming, since the sizes are actually controlled by compiler.

shigutso, 2022-10-21:
> But if no one contacts me in discord

Sorry, I can't find the Discord invite link in this repo, would you mind sharing? Thank you

CTCaer, 2022-10-21:
<https://discord.gg/erNC4FB>

CTCaer, 2022-10-21:
So the rcmloader thing is a FAT driver bug on these injectors.
Mitigated if you format the drive as FAT16 with 512 clusters and copy paste the folders and payloads back.

isermann, 2022-10-21:
@CTCaer I reached out to you in Discord but you seem to be offline?

CTCaer, 2022-10-21:
Just checking the list if I'm online is not "reaching out" btw.
(Try to write something and PM it or ping me)

CTCaer, 2022-12-18:
Closing as a non-hekate issue

closed 2022-12-18 by CTCaer