CTCaer/hekate #787: Hekate 5.9.0 not possible to mount eMMC boot0 / boot1 over UMS

dezem, 2022-10-16:
With hekate 5.9.0 it's not possible to mount the boot0 / boot1 from eMMC.
If i downgrade to hekate 5.8.0, there are  no problems.

[image]

CTCaer, 2022-10-17:
Confirmed. Btw all eMMC modes are broken. GPP/BOOT0/1. Suprised you reach that far and not hang in `Mounting disk`.
Silly mistake on refactoring:
[image]

For now you can use the 1.3.0 `nyx.bin` (from v5.8.0).
(Changing only Nyx to an older version, does not affect booting HOS 15.0.0 via hekate, aka fss0.)

Thanks for reporting it. Will be fixed in next release.
(If you really need anything from the v1.4.0 Nyx features, tell me and I will compile a binary and attach it here before.)

dezem, 2022-10-17:
It will be enough in the next release 😉 

Thanks so far!

CTCaer, 2022-12-20:
Fixed in latest

closed 2022-12-20 by CTCaer