CTCaer/hekate #790: Failed to backup RAW GPP/ hekate 5.9.0/ Nyx 1.4.0

Richmong97, 2022-10-20:
Hello, I have a problem when making a backup in RAW GPP and they are these errors "fatal error when writing to the SD" and "Error writing 8192 blocks \@LBA" I investigated and found that the same problem occurred on June 29, 2019, the exact same thing happens to me, is there a way to fix it? I will really appreciate the help (Sorry for my English, I'm using a translator)

Richmong97, 2022-10-20:
the information requested @Masamune3210 
[image]

Masamune3210, 2022-10-20:
You submitted the comment too fast, the picture didn't finish uploading

Richmong97, 2022-10-20:
[image]

Richmong97, 2022-10-20:
I'm not sure if it's a problem with the SD card or a bug with my console @Masamune3210

CTCaer, 2022-10-21:
Your issue seems to be broken pin on sd reader - since your sd is legit.
(That, assuming you don't have corruption on your filesystem)

You should retry to backup and when that error happens, go into microsd info and take a screenshot again. Without reloading hekate/Nyx.
The screenshot will show what it is:
- 0 errors is fs corruption
- errors is sd reader issue.

Richmong97, 2022-10-21:
A few minutes ago I started doing everything again but as soon as I tried to start hekate I got an error indicating a problem with the SD card reader Reboot the device and it didn't recognize the SD card, so reboot the device but this time pressing on the SD card slot It recognized the SD card without any problem, so I decided to start hekate (always pressing on the slot) Did BOOT0 AND BOOT1 without any problem, then I started RAW GPP and everything was fine until I decided to remove my finger from the afternoon SD slot, as soon as I did it the error appeared. I'll post the photos below.

Richmong97, 2022-10-21:
[image]
[image]

Masamune3210, 2022-10-21:
You might be able to get away with replacing the SD card reader board , but if the pin is truly broken or there is a issue with the connector, nothing is going to be able to fix it short of a solder job

Richmong97, 2022-10-21:
I really appreciate your help, thanks for everything and for helping me solve the problem

closed 2022-10-21 by CTCaer