CTCaer/hekate #791: v5.9.0 hangs sometimes while initializing

TSRBerry, 2022-10-29:
The current version sometimes forces me to reinject the payload because it hangs while it's initializing. This affects reboot2payload as well as normal injection via USB.

When it hangs I see the hekate menu screen, but the battery will show up as 0% and date + time stay at 0. The joycons haven't vibrated yet and I'm forced to power the console off by holding the power button for about 6 seconds.

If I start reinjecting after that I always run into the same issue, but if I wait a few seconds (~30) I can inject the payload successfully and hekate starts normally.

I never had these issues before, so I suspect it might be a new issue with v5.9.0.

Let me know if I can help you somehow to fix this issue.

CTCaer, 2022-10-29:
Redownload hekate from here and report back.

TSRBerry, 2022-10-29:
No idea what changed, but I redownloaded and extracted it and was able to inject the payload successfully on my first try. Reboot2Payload also worked fine.

Was this file corruption? I was initially guessing it couldn't be that, because it shouldn't even be able to execute in the first place if that was the case.

CTCaer, 2022-10-29:
File corruption or you didn't get hekate from here.

closed 2022-10-29 by CTCaer

TSRBerry, 2022-10-29:
File corruption it is then. I only get it directly from the releases here