CTCaer/hekate #792: OLED 15.0.1, Can't boot OFW when both are 15.0.1

vegedb, 2022-11-03:
Using latest Atmosphere and Hekate.

Method of booting to OFW [Hekate Reboot menu>OFW]

==================================================
Before, mismatch firmwares
Emunand = 15.0.0 [WORKING]
OFW = 15.0.1 [WORKING]

==================================================
After updating Emunand to 15.0.1 ONLY,
Emunand = **15.0.1** [WORKING] 
OFW = 15.0.1 [NOT WORKING]

==================================================

### Explaination
Before, having mismatch firmwares surprisingly worked without any problems or so I thought. Seems like it affected the chip to take longer to glitch. So I decided to match the firmware.

After updating emunand using Daybreak from 15.0.0 to 15.0.1 to have matching firmwares, OFW doesn't boot anymore, it shows black screen or dead no signs of life. To reboot, holding power button for a few seconds is needed.

Weird thing is, only Emunand works fine.

closed 2022-11-03 by vegedb

TheSpyKid0906, 2022-11-28:
Hey! I'm looking to mod my switch in 15.0.1. Can you use hekate? Or did you restore back with no emulator? I don't know but don't want to corrupt anything myself if you already had a problem with it. Thanks!

Masamune3210, 2022-11-29:
take a nand backup, always take a nand backup before doing anything that has a chance of causing the system to become dysfunctional. If you run into issues, restore the nand backup. There is very little that can be done to a brom-exploitable switch software wise that cant be fixed by just restoring a backup