CTCaer/hekate #793: hello eveyone!

ztyjason, 2022-11-12:
hello,my name is jason,from china.
please forgive my bad english.
I have a very troublesome problem.
I used these few tools to restore the switch.
Tools:EmmcHaccGen.2.2.3    HacDiskMount    Hekate5.9.0

I have used these tools and recovered many switch.
Was successful before FW15
but,it has been unavailable recently.
warning : pkg2 read failed ! failed to launch HOS
I checked a lot of information, but nothing works.
Hope you all give some advice.
thank you very much !!!

CTCaer, 2022-11-12:
Basically it can't find the `BCPKG2-1-Normal-Main` partition.
So your GPT table in eMMC is corrupt or missing.
You will need to fix that.
Check https://switchbrew.org/wiki/Flash_Filesystem#User_Partitions these will need to exist there.

urherenow, 2022-11-12:
Github isn't a help forum. FW15 has new keys you probably haven't dumped yet. The latest Hekate needs to be paired with the latest Atmosphere.

ztyjason, 2022-11-13:
> 

Thank you for your reply.
I've tried fixing the GPT list, but that doesn't work.
I don't know what is causing the EMMC damage.
Could EmmcHaccGen.2.2.3 be his reason, but I succeeded in writing fuses 15.
HacDiskMount???

ztyjason, 2022-11-13:
> Basically it can't find the `BCPKG2-1-Normal-Main` partition. So your GPT table in eMMC is corrupt or missing. You will need to fix that. Check https://switchbrew.org/wiki/Flash_Filesystem#User_Partitions these will need to exist there.
> 
> 基本上它找不到 BCPKG2-1-Normal-Main 分区，因此 eMMC 中的 GPT 表已损坏或丢失。你需要解决这个问题。检查这些 https://switchbrew.org/wiki/flash_filesystem#user_partitions 是否需要存在。

I use my switch to write and it also fails.
"read pkg2 failed" appears
Can it be repaired using donor PKG2?

ztyjason, 2022-11-16:
> Basically it can't find the `BCPKG2-1-Normal-Main` partition. So your GPT table in eMMC is corrupt or missing. You will need to fix that. Check https://switchbrew.org/wiki/Flash_Filesystem#User_Partitions these will need to exist there.

Probably the reason for EmmcHaccGen.2.2.3.
Because the pkg2 it makes is broken.
thank you for your help.

closed 2022-11-21 by CTCaer