CTCaer/hekate #795: fss0 package 3 error message

ScytheMeister7, 2022-11-12:
Whenever I try to launch atmosphere it tells me that I have a wrong ini cfg or missing/corrupt files and when I looked in the ini folder it was empty and I'm not sure what to put there or how to fix it, and I'm dumber than a rock when it comes to stuff like this could I get some help?

Masamune3210, 2022-11-12:
You need to make a config file, there should be examples somewhere in the files of the repo or in the bootloader folder iirc

CTCaer, 2022-11-12:
You also need to extract atmosphere in your sd's root folder.
It tells you what file is missing/corrupt. And that message means you either missing it or it can't load it.

closed 2022-11-12 by CTCaer

ScytheMeister7, 2022-11-12:
It says it's package three that's missing or corrupt but it's not missing and I've already tried replacing it

> You also need to extract atmosphere in your sd's root folder. It tells you what file is missing/corrupt. And that message means you either missing it or it can't load it.

Okay I extracted it, and it worked. Thank you!! Sorry for having such a simple problem lol