CTCaer/hekate #798: Stuck in EmuMMC

Anuskuss, 2022-11-23:
This is what I did:
- Grabbed Atmosphère v1.4.0 and put the files on the SD
- Grabbed hekate v5.9.0 and put the files on the SD
- Run the exploit and selected "CFW (emuMMC)"
- Install the firmware update

And now I'm stuck in EmuMMC, meaning I don't see the boot menu and it just goes straight to EmuMMC. Autoboot is off and I haven't touched any other file other than overwriting the preexisting files with the updated ones. I think I either messed something up or the new version just doesn't offer a boot menu? Everything else works fine and I can still get into Stock by turning the Switch off and not running the exploit.

```ini
[config]
autoboot=0
autoboot_list=0
bootwait=3
backlight=100
autohosoff=0
autonogc=1
updater2p=0
bootprotect=0

[CFW (sysMMC)]
emummc_force_disable=1
fss0=atmosphere/package3
icon=bootloader/res/icon_payload.bmp

[CFW (emuMMC)]
emummcforce=1
fss0=atmosphere/package3
icon=bootloader/res/icon_payload.bmp
payload=bootloader/payloads/fusee.bin

[Stock]
emummc_force_disable=1
fss0=atmosphere/package3
icon=bootloader/res/icon_switch.bmp
stock=1

[Fusee]
icon=bootloader/res/icon_payload.bmp
payload=bootloader/payloads/fusee.bin
```

Sorry I'm a bit of a noob and I forgot most things I did in the past because I touch the Switch maybe once a year.

impeeza, 2022-11-23:
1) This space is not a support site, please go to Discord or GBATemp
2) please read the instructions of use of Hekate on main page, when you inject payload, keep pressed Vol- and you will get into NYX. then you can setup your environment.

Anuskuss, 2022-11-23:
Yeah, I obviously tried injecting the payload / rebooting and holding `Vol-` but that didn't work. The thing is that before the upgrade it used to just boot into the boot menu (NYX?) and I didn't have to press anything, now it just skips it and I can't reenter. What am I missing? I could also ask on Discord if you don't want to help me further.

Anuskuss, 2022-11-23:
Okay I found the issue: I've been sending `fusee.bin` to my Switch for ages but if I instead send `hekate_ctcaer_5.9.0.bin` it works like it used to. Hope this info helps somebody someday.

closed 2022-11-23 by Anuskuss