CTCaer/hekate #800: is it possible to put evreything on the emmc (including hekate)

hannescam, 2022-12-18:
is it possible to put evreything on the emmc (including hekate) without using any microsd and i dont care about hos (i have a backup)

CTCaer, 2022-12-18:
Yes and no.

No as it is now.
Yes if you edit the relevant files (sd.c, diskio.c, etc), compile and copy them to a fat partition into eMMC.

Everything linux based can be booted or run from eMMC. (For HOS CFW, sd is mandatory.)

Replying simply to your question as it is, so basically if it's possible out of the box, no.

closed 2022-12-18 by CTCaer

hannescam, 2022-12-18:
@CTCaer what modifications do i have to make (i can programm c and c++)
and do i need to change/modify the part that i need to inject

CTCaer, 2022-12-20:
Start with the objects I referred above (sd.c and diskio.c) and refer to emmc.c also.

hannescam, 2022-12-21:
where are these located and what modifications i need to make (i dont need nyx) to boot it into ubuntu completley without emmc (l4t is allready on emmc)