CTCaer/hekate #801: When formatting microSD cards take SDA's recommendations into account.

profi200, 2022-12-22:
From a quick glance i could not tell if hekate does take the recommendations from "SD Specification Part 2 File System Specification" (Can't link it here. Search for it.) into account. While they do recommend exFAT only for SDXC cards i got improved random read/write by applying the recommendations for FAT32 on SDXC cards as well. Also see my (unfinished) [tool](https://github.com/profi200/sdFormatLinux).

Benchmarks (ignore the SSD one): https://www.dropbox.com/sh/cz8i1z4nq4tvqo5/AADQMv8xn9U1icis0tA5asfca/benchmarks

impeeza, 2022-12-22:
https://www.openimpulse.com/blog/wp-content/uploads/wpsc/downloadables/SD-Card-File-System-Specifications-.pdf but is marked as confidential ;(

profi200, 2022-12-22:
Well, hence why i said i can't link it here.

CTCaer, 2022-12-22:
It does and then more.
Like day 1 of the partition manager release

closed 2022-12-22 by CTCaer

profi200, 2022-12-22:
Good to hear. Will have a deeper look at what your partition manager does when i have time.