CTCaer/hekate #802: RCM Loader ONE Plus (White Box) does not inject payload.bin Hekate 6.0.1

joaomeuputo, 2022-12-23:
Hello, after a few tests with my RCMloader ONE Plus (already upgraded firmware version to v2.02), I did notice that Hekate 6.0.1 payload does not work with RCM Loader.

Below are the following Hekate versions and their payload state:
5.6.1 (Pre-installed with RCM Loader) - WORKS
5.9.0 - WORKS
6.0.0 - WORKS
6.0.1 - DOESN'T WORK

Thanks in advance, and keep doing a great job as usual!

Merry Xmas.

urherenow, 2022-12-23:
Then use 6.0.0 in your loader. It should not effect anything at all, because if /bootloader/update.bin is a newer version, your switch will use that one anyway. No need to keep updating your RCMloader.

CTCaer, 2022-12-23:
As I said before that's a bug on rcm loader firmware. You can format the drive to FAT16 with 512B cluster and recopy stuff back.

closed 2022-12-23 by CTCaer

nefariouslegion, 2023-01-08:
How do you go about formatting it to FAT16? I seem to only have the option for NTFS or FAT12

urherenow, 2023-01-08:
https://www.sdcard.org/downloads/formatter/

nefariouslegion, 2023-01-08:
> https://www.sdcard.org/downloads/formatter/

That still formats it to FAT12 instead of FAT16 as you can see in the screenshot 
[image]

Masamune3210, 2023-01-08:
format *: /FS:FAT /Q
where * is the drive letter assigned to the sd card partition

nefariouslegion, 2023-01-08:
> format *: /FS:FAT /Q where * is the drive letter assigned to the sd card partition

Tried that prior to the rcmloader getting bricked and it just said "Invalid drive specification" even though the drive letter was correct

urherenow, 2023-01-08:
Because it's a fake, damaged or both, and doesn't even have 1MB capacity. In theory, fat16 should have 4.3MB capacity (4.1MB formatted). My genuine RCMLoader has a 2MB capacity. Yours is only reporting 712kb? I'll just go back to my original reasponse: Keep 6.0.0 on your loader and forget about it. You're not saving any time by updating the loader, because /bootloader/update.bin is still going to be read/checked every time anyway.

impeeza, 2023-01-08:
please do not use Issue report as support page.  Use Discord or GBATemp for that.

FAT 12 and FAT 16 are used by the formatting utilities indistinctly as the disk is < 2Mb then the utilities always use FAT12

On GBATemp there is a thread about these RCM Gadgets:

https://gbatemp.net/threads/fake-rcmloader-dongles-in-the-wild.620470/post-10024763

please close this issue as is not a Hekate issue.

CTCaer, 2023-01-08:
@urherenow You basically caused an issue because you are misinformed.
Such devices rely on SFD partition scheme..
That tool is only for normal drives. Not tiny ones.

And FAT12 is also fine. The important thing is the 512 cluster size. The rest are irrelevant.

Anyway, @nefariouslegion you should see if the driver is there and not mounted. If it's not, you should search google on how to make it show up (via fw update or something else).

nefariouslegion, 2023-01-08:
> @urherenow You basically caused an issue because you are misinformed. Such devices rely on SFD partition scheme.. That tool is only for normal drives. Not tiny ones.
>
> And FAT12 is also fine. The [...]

I don't think the issue is it not being mounted. It no longer lights up at all whether plugged in or not. So I think the whole thing is bricked. Though oddly it was still fine after running that tool until I removed the loader from the PC. Then it suddenly didn't work.

urherenow, 2023-01-08:
Sorry. Deleted my other comment to avoid future confusion when others might check for solutions to this problem, but it's showing in OP's reply. But to be fair, I did initially say to leave 6.0.0 on and and to leave it alone. You are the one who suggested trying to format to fat16.

CTCaer, 2023-01-09:
@nefariouslegion Tell the vendor that you formatted it, via windows format and doesn't work now. Don't tell them exactly how.
And ask them how to show the storage on pc to reformat it with another tool. And tell them that bootloader mode works, since you can fw update.

@urherenow I said to format it. Not partition it, which is what sd formatter does. It uses MBR scheme and also aligns the partition.
These amtel chip's firmware do not support proper partition schemes. They expect SFD.
And as it seems that they don't have a fallback mode into UMS if they fail to read it.

urherenow, 2023-01-09:
If it’s actually a One Plus, then files are mirrored here for fw upgrade: https://gbatemp.net/download/rcm-loader-one-plus-firmware-v2-02.37758/

nefariouslegion, 2023-01-09:
The firmware isn’t what I need. I already have that and like I said the RCMLoader is detected in firmware upgrade mode. So I ran the updater hoping it would un-brick it. It ran and said “Pass” but then still remained bricked when not in firmware upgrade mode

impeeza, 2023-01-09:
Seriusly, this is not a suppor site, read the page on gbatemp, you bricked you loader beyond recovery, because you are a Fake one, please close the issue is not related to Hekate.

https://gbatemp.net/threads/fake-rcmloader-dongles-in-the-wild.620470/

nefariouslegion, 2023-01-09:
Except everything I can find online points to it being real. You’re just assuming it’s fake with almost no evidence. You haven’t even seen it.

https://gbatemp.net/threads/fake-rcmloader-dongles-in-the-wild.620470/<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgbatemp.net%2Fthreads%2Ffake-rcmloader-dongles-in-the-wild.620470%2F&data=05%7C01%7C%7C7d91b5b571a841478fc808daf1fb7cbd%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C638088359674441772%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=zRMHVeiMjyCQHDqYHFenc3wDp1w%2Ft9Qc0UC96s5F0Bc%3D&reserved=0>

urherenow, 2023-01-09:
You couldn’t inject 6.0.1. That’s evidence enough. Want more? You show less than 1MB capacity. It’s a clone.