CTCaer/hekate #804: Unable to boot into Hekate 6.0.1 with RCMLoader

Danfun64, 2022-12-25:
For whatever reason, Hekate refuses to boot into Hekate 6.0.1 directly from the firmware file through RCMLoader. However, I am able to load 6.0.1 from the bootloader directory in my SD card when I use the 6.0.0 firmware in RCMLoader.

urherenow, 2022-12-25:
This has already been posted and closed. Read the damn release notes before downloading and you’d know that you don’t need to update your RCMloader. Put 6.0 on it and leave it alone. You will always load the most recent from /bootloader/update.bin.

impeeza, 2022-12-25:
An old case of RTFM.  

This is duplicate of https://github.com/CTCaer/hekate/issues/802

Please if you need help or support go to the proper site. Discord and gbatemp are examples. This space is for hekate issues only

More info on the posts:
https://gbatemp.net/threads/fake-rcmloader-dongles-in-the-wild.620470/post-10024785

https://gbatemp.net/threads/fake-rcmloader-dongles-in-the-wild.620470/post-10031676

closed 2022-12-25 by CTCaer