CTCaer/hekate #805: NYC exception occurred (LR 810646AC) UNDEF

fabfr91, 2022-12-25:
With the following releases, when I launch hekate through TegraRcmGUI, I meet the yellow error message "NYC exception occurred (LR 810646AC) UNDEF"
- hekate v5.9.0 & Nyx v1.4.0
- hekate v6.0.0 & Nyx v1.5.0
- hekate v6.0.1 & Nyx v1.5.1

No problem with hekate v5.8.0 & Nyx v1.3.0

Config : Switch V1 unpatched, FW 15.0.1

impeeza, 2022-12-25:
I can confirm what TegraRCMGUI on Windows 11 with latest Drivers works fine with hekate 5.9.0, 6.0.0 and 6.0.1, Erista V1 console works fine:

```
RCM Device detected
Invoking TegraRcmSmash.exe with args : "H:\hekate_ctcaer_5.9.0.bin"
TegraRcmSmash (64bit) 1.2.1-3 by rajkosto
Opened USB device path \\?\usb#vid_0955&pid_7321#6&1c446172&e&1#{50e5e553-1957-48ba-8b36-79da4affe0ba}
RCM Device with id 8083020A00000004C6A2406401101062 initialized successfully!
Uploading payload (mezzo size: 92, user size: 105953, total size: 172169, total padded size: 176128)...
Smashing the stack!
Smashed the stack with a 0x7000 byte SETUP request!
Payload successfully injected
RCM device disconnected

RCM Device detected
Invoking TegraRcmSmash.exe with args : "H:\hekate_ctcaer_6.0.0.bin"
TegraRcmSmash (64bit) 1.2.1-3 by rajkosto
Opened USB device path \\?\usb#vid_0955&pid_7321#6&1c446172&e&1#{50e5e553-1957-48ba-8b36-79da4affe0ba}
RCM Device with id 8083020A00000004C6A2406401101062 initialized successfully!
Uploading payload (mezzo size: 92, user size: 109056, total size: 175272, total padded size: 176128)...
Smashing the stack!
Smashed the stack with a 0x7000 byte SETUP request!
Payload successfully injected
RCM device disconnected

RCM Device detected
Invoking TegraRcmSmash.exe with args : "H:\hekate_ctcaer_6.0.1.bin"
TegraRcmSmash (64bit) 1.2.1-3 by rajkosto
Opened USB device path \\?\usb#vid_0955&pid_7321#6&1c446172&e&1#{50e5e553-1957-48ba-8b36-79da4affe0ba}
RCM Device with id 8083020A00000004C6A2406401101062 initialized successfully!
Uploading payload (mezzo size: 92, user size: 109080, total size: 175296, total padded size: 176128)...
Smashing the stack!
Smashed the stack with a 0x7000 byte SETUP request!
Payload successfully injected
RCM device disconnected

```

please for support go to the adequate sites like discord and GBATemp, you could have a bad USB Cable or driver error.

CTCaer, 2022-12-25:
You have a fake or corrupted sd

closed 2022-12-25 by CTCaer