CTCaer/hekate #809: Blank boxes after first boot

Frosty3001, 2022-12-30:
This is my first time with Switch emulation so I'm being as cautious as possible with the process. I injected hekate_ctcaer_6.0.1.bin and TegraRcmGUI showed "RCM O.K." but there's no text in the boxes on the Switch. This isn't my own photo but this is exactly what it looks like on my end.
[image]
I've searched everywhere online for an answer but none of the images showed blank boxes like this one.

impeeza, 2022-12-30:
Its mire likely what you are missing some files on your sd card or the folder RED inside the BOOTLOADER one

Frosty3001, 2022-12-30:
> Its mire likely what you are missing some files on your sd card or the folder RED inside the BOOTLOADER one

Did you mean the res folder? I have a "res" folder but not a "red" folder.

impeeza, 2022-12-30:
Yes res. Sorry a typo. And remember to check the content of the res folder

Frosty3001, 2022-12-30:
I have icon_payload.bmp and icon_switch.bmp both at 145KB

CTCaer, 2022-12-30:
You have corrupted files or filesystem.
Backup your hekate_ipl.ini and the inis, assuming they are not corrupted, delete bootloader folder and reextract the release zip.

closed 2022-12-30 by CTCaer