CTCaer/hekate #813: Nyx exception occured (LR 810648A0) UNDEF

dnsBlah, 2023-01-03:
Receiving this error message on Hekate 5.9.0, 6.0.0, 6.0.1 releases.
Unpatched V1 erista model, HOS version 15.0.1

Tried multiple Sandisks (confirmed in Console Info)
FAT32, 32kb cluster size.
All formatted using guiformat on Windows.

Not sure how to resolve this or where it comes from. Another friend had same like issue randomly and random freezing. Was unable to create eMMC sys backups.
After unpairing all controllers in HOS it worked fine.

But for this one it still does not work.

The error comes up randomly and when pressing a key it goes back to the main screen.
If you need any info, for sure I will try to provide it asap.

urherenow, 2023-01-03:
You have a corrupted folder or file somewhere on your sd. From a fresh format, try to extract ONLY the latest hekate and Atmosphere, and restore ONLY your ini files, and see if things boot up to HOS. Then you can add more stuff back a bit at a time until you either reproduce the error, or perhaps if you're lucky the error will correct itself by doing these steps.

I'll assume you've tested your card with h2testw. And better yet, you may want to post a screenshot of your sd info from nyx here, if you still have issues.

dnsBlah, 2023-01-03:
[image]

Actually only have bootloader on at the moment I believe. I instructed my mate to only put Hekate on after formatting with guiformat to fat32 with 32kb cluster size.
Tried 3 releases from Hekate.
We're not even able to boot yet, the exception comes fast and goes back to home screen after pressing a button.

Is there any logging that can be enabled for nyx?

urherenow, 2023-01-03:
If boot loader is all that’s on there (you really should have atmosphere as well), then format it within nyx. ‘Bout to sleep, but off the top of my head, it might be in in the partition part of nyx to format it. (Partition the card, as 1 fat32 partition. Iirc, nyx will make it a 64k cluster.

CTCaer, 2023-01-06:
First of all do not use guiformat.
Secondly, your reporting on exception is useless, since these "weird" numbers are tied with a version. So you need to use a specific version (e.g. 6.0.1 hekate and 1.5.1 nyx) and tell me the number and where.

That is normally sd corruption or hw issues.
So unless you do a proper testing and report, I can't tell you why it happens.

dnsBlah, 2023-01-06:
alright. asking for having all files deleted, new clean download of 6.0.1 release,  a new sd info screenshot,
format sd through nyx's partition manager, and new sd info screenshot. 
If it he gets that far.

Will be back soon. You're much appreciated!

dnsBlah, 2023-01-06:
So having some discussions on how to prepare the SD properly for Hekate tho.
What is your advised method to prepare the SD?

On Hekate 6.0.1 wirh Nyx 1.5.1 from latest release, the error shows randomly.
Sometimes straight away on boot, otherwise it takes few seconds.
BUT when opening partition manager it goes into exception immediately.
So he can't format using Hekate at the moment. Asked him to format to exfat as that is a option available on his windows out of the box and try again to go through Nyx's partition manager.

dnsBlah, 2023-01-06:
[image]

dnsBlah, 2023-01-06:
[image]

[image]

impeeza, 2023-01-06:
@CTCaer Hi, first at all, happy new year, sorry for bother, what is the problem when you use GuiFormat?  Thanks for all your work.

dnsBlah, 2023-01-06:
From exFat partition, yet same behaviour with the error. After numerous times could get through the partition manager and made it a fat32 64kb clustersize. Copy hekate 6.0.1 + nyx1.5.1 again, still gives the same exception.
[image]

CTCaer, 2023-01-06:
Number seems always the same which is good, it's blurry though.
it's `81064b64`?

dnsBlah, 2023-01-07:
Thanks for your recommendation on the formatting tool!

The Nyx exception number is 81064B64
The images got little blurry indeed, my apologies

CTCaer, 2023-01-07:
Well that address doesn't have code so that's weird.
I would suggest to open your bootloader/nyx.ini and edit `bpmpclock=` to `bpmpclock=2` and see if that fixes it.

dnsBlah, 2023-01-07:
bpmpclock already was on 2. Tried 0 and 1 but all same behaviour.
Any more idea's to test?

CTCaer, 2023-01-07:
Seems like a hw issue then. Send a PM in discord. (Find my tag via retronx server.)

dnsBlah, 2023-01-07:
during version rolldown tests, Nyx v1.3.0 does not show this exception error

dnsBlah, 2023-01-08:
Using nyx release from hekate5.8.0 it worked better, but got different exception.
Thanks for your valuable time on discord it was a matter of the bpmpclock to be hardcoded to 88.

Thanks again!

CTCaer, 2023-01-08:
Next update will have a mode for crap SoC binnings.

closed 2023-01-08 by CTCaer