CTCaer/hekate #814: When using RCMLoader with Hekate 6.0.1 it just results in a blank screen

nefariouslegion, 2023-01-04:
Hi,
If I put hekate 6.0.1 on my RCMLoader it does not work. I get the blue light on the loader as if it's attempting to load the payload but then it just stays at a blank screen. The same file works fine when injecting from PC with tegra.

However if I downgrade to hekate 5.9.0 on the RCMLoader it works fine and loads up 6.0.1 since it's on my SD card. But I'm wondering how to fix this?

urherenow, 2023-01-04:
Please look at the closed issues. This has been asked and answered too many times already. Bottom line, leave 6.0.0 on your dongle, and leave it alone. You’ll still be using the latest anyway, because that’s what bootloader/update.bin is.

CTCaer, 2023-01-06:
Again that's a bug on the rcmloader's fat/fs driver.
https://github.com/CTCaer/hekate/issues/802#issuecomment-1363990676
If you actually want it fixed, ask the manufacturer for a firmware that remedies the issue.

closed 2023-01-06 by CTCaer