CTCaer/hekate #815: When i launch stock Sysnand, it hangs on Nintendo logo

favoulos, 2023-01-05:
Ss per title, when i try to start my stock sysnand (3.0.2) from Hekate, it hangs on nintendo switch logo. I tried disabling RCM mode and it starts fine, so it's not a nand problem. If I try to start it "non stock" with atmosphere, it starts.

This is my hekate config:
[Stock - sysMMC]
fss0=atmosphere/package3
stock=1
emummc_force_disable=1

Does the stock option support firmware 3.0.2? Do you have any ideas?
OT: Do you have any discord or reddit servers where we can discuss?

WVoo, 2023-01-05:
I can confirm this problem, it also occur on my switches FW 2.3.0 and FW 1.0.0
Upgraded from 5.5.8 where everything worked without problems.
Same config as above.

CTCaer, 2023-01-06:
You can contact me with a PM via the retronx discord.

Aruno4, 2023-01-22:
Greetings, this is unrelated to OP's problem but since using the payload provided on the fix solved my problem might as well mention it here.

After updating emuNAND/AMS/Hekate to the latest releases available, my RCM injector seemed to not work after updating it with the latest Hekate payload (the LED of the injector was booting normally and I could connect it via USB to transfer files but after plugging it to the switch after entering RCM, Hekate didn't boot). However, after replacing the payload with the one provided in the fix (or even using the v5.9.0 ) it started working again.

Also worth mentioning, during the time I was not able to use the injector I used Rekado/Tegra to inject the same payload and via those 2 options Hekaye booted as intended.

CTCaer, 2023-02-23:
The stock fix was added in release

closed 2023-02-23 by CTCaer