CTCaer/hekate #819: error while loading fss0 : atmosphere/package3

sertose, 2023-01-30:
I installed the most recent versions of atmosphere (1.4.1) and hetake (6.0.1), when I enter RCM and start the hetake payload, I go to launch and start Atmosphere FSS0 EmuMMC, I get a red message: Error while loadingg fss0: atmosphere/package3 wrong ini cfg or missing/corrupt files !
I made sure that the hetake file ends with .ini and not with txt and in the code the entry is fss0=atmosphere/package3
can someone tell me what i'm doing wrong?

Masamune3210, 2023-01-30:
Try wiping atmosphere from the sd card and copying it over fresh. Chances are a file is corrupted somewhere and its causing your issue

sertose, 2023-01-30:
I downloaded the zip atmosphere again (1.4.1)
but it keeps giving me error.
I don't know if the error is from an atmosphere file or from the hetale_ipl.ini code

impeeza, 2023-01-31:
Please remember this is a issue report site, not a support forum, please go to Discord or GBATemp for ask support, on GBATemp you can show your Hekate_ipl.ini code and somebody will try to help.

is for sure you Hekate_ipl.ini file have a error.

please close this because do not appears be a issue.