CTCaer/hekate #821: [bug] Reboot to OFW doesn't work for AutoRCM

Yoti, 2023-02-08:
It goes to the RCM mode instead of OFW.
I suggest disable or add a note to the item.

urherenow, 2023-02-08:
And I suggest you learn what AutoRCM does. This isn’t a bug.

Yoti, 2023-02-08:
@urherenow it is. Nyx can check for the AutoRCM state and can disable this button if it's On. Also, I told this to CTCaer early and he asked to open an issue IIRC.

Masamune3210, 2023-02-08:
Its a bug that it lets you try, knowing that it will fail. Its not a bug that it fails

urherenow, 2023-02-09:
@Yoti, It isn’t a simple matter of “disabling a button” for RCM. Again, learn what AutoRCM does. I get the feeling that disabling the “Reboot to OFW” button is what will be done here (for cases where AutoRCM is detected).

Masamune3210, 2023-02-09:
Why are you being so malicious? Also, I'm pretty sure that's what they are talking about disabling, making it so that the Reboot to OFW button is disabled if AutoRCM is detected as enabled

CTCaer, 2023-02-23:
Behavior changed in latest

closed 2023-02-23 by CTCaer