CTCaer/hekate #822: Error de inicio de hekate_ctcaer_6.0.1_Nyx_1.5.1

Hugo7801, 2023-02-16:
cuando inicio hekate me sale un error.
Es una tarjeta nueva (de un 1t de xiaomi) me sale este error siempre lo e intentado varias veces y nada esta en exFat.
Antes tenia una trajeta (de 32g de SamDisK) y también me salía ese error pero intentándolo mucho ya por fin me funcionaba.
Los escribo el mensaje de error.
Missing LP0 <sleep> lib!
Missing Minerva lib!
Update bootloader folder!
Press any key...

Espero que me contestéis por favor.

urherenow, 2023-02-16:
That isn't a bug, and this isn't a help forum. Properly extract the entire .zip file to your sd card. The error clearly tells you this.

impeeza, 2023-02-16:
Hola 

Primero que todo recuerda que este es un espacio únicamente para reportar errores o hacer solicitudes de méjora.  no es un foro de soporte.

Para soporte debes dirigirte a los sitios correctos, como por ejemplo, Discord y gbatem.net.

en ellos te dirán que seguramente es uno de los dos siguientes problemas:

Tienes una tarjeta falsa o defectuosa, utiliza H2testW (https://gbatemp.net/threads/my-ams-emulator-cant-run-games.615664/post-9885418).

O estás descomprimiendo mal los archivos,  descarga NUEVAMENTE desde el repositorio GitHub tanto Hekate como Atmosphère y descomprime el contenido de los archivos ZIP en la CARPETA RAÍZ de tu tarjeta SD.

Si vas a utilizar GBATemp.net recuerda que es un sitio únicamente en inglés.

impeeza, 2023-02-16:
por favor cierra este requerimiento ya que no es un reporte de falla.

closed 2023-02-17 by Hugo7801