CTCaer/hekate #824: Can't Dock and play on tv.

Aminandri, 2023-02-20:
Hello,
i' m new here i have a question, 3 years ago i used to play with a cfw with atmosphere my games on tv my switch just use to dock normally. 2 days ago i did everything right,i put atmosphere everything, i installed some NSP and everything was fine but when iput my console on the dock nothing hapeend it doesn't show anything.
I have tow consoles i tried the other one and the dock which is brand new works e thi is the fun part when i put my cfw console with original fw it shows up but with cfw not.
Now my question is this i want to play as i used to with my cfw in docked mode on tv and ican't, can someone help me or explain it to me? 
Sorry for my ignorance and my English i hope is understandable.
Thank you.

urherenow, 2023-02-20:
Check your TV settings on cfw in handheld mode. Resolution or any other number of settings might not be compatible with your tv. But this is not a support forum. You should use discord or gbatemp to ask for help.

Aminandri, 2023-02-20:
I' msorry i ddnt new where to write.
I already ryed to chenge settings and all my question is: is still possible toplay docked mod?

CTCaer, 2023-02-20:
Piracy not supported
Also hekate has nothing to do with docking or anything at that matter.

closed 2023-02-20 by CTCaer