CTCaer/hekate #826: Hekate Update 16.0

bdawg1989, 2023-02-23:
Hello,

Now that Atmosphere is pre-released and working with the new 16.0 switch update, what is the ETA on a new update for Hekate?

North-West-Wind, 2023-02-23:
While we wait for that, is there a way to boot into sysMMC without Hekate?

TTghostsemir, 2023-02-23:
Yes with the Fusee.bin

warnooob, 2023-02-23:
it won't boot into syscfw with the fusee.bin unless you don't have an emummc in the first place

North-West-Wind, 2023-02-23:
@warnooob I was trying to figure it out and I did. You have to disable emuMMC in `emuMMC/emummc.ini`. Change `enabled=1` to `enabled=0`

NudelsHelp, 2023-02-23:
@warnooob How???

iero19, 2023-02-23:
Is this pkg1 error because of Hekate not being updated yet? https://imgur.com/a/Law0Oyz

warnooob, 2023-02-23:
@iero19 yes, use fusee for now while you wait for hekate to update for 16.0.0. remember to delete your sysmodules and themes too, both will result in atmosphere getting an error on boot

iero19, 2023-02-23:
Got it. I have no idea what sysmodules are and I do not have any themes installed, so I think I'm safe in that aspect. Is there a way to boot into sysMMC for now?

NudelsHelp, 2023-02-23:
@warnooob how to Delete them?

North-West-Wind, 2023-02-23:
If that's the case, you can add that section manually:
```ini
[Fusee]
icon=bootloader/res/icon_payload.bmp
payload=bootloader/payloads/fusee.bin
```

CTCaer, 2023-02-23:
New version released

closed 2023-02-23 by CTCaer

NudelsHelp, 2023-02-23:
Okay but i Need to have the Pre Release of Atmosphere or the 1.4.1?

North-West-Wind, 2023-02-23:
Pre release (1.5.0). 1.4.1 doesn't work with switch 16.0.0.