CTCaer/hekate #831: Bootlogos Graphical Glitch

Bladen123, 2023-02-23:
Hey CTCaer 

Splashscreens  do graphical glitches like a pc game without VSync , it splits the picture for a 0.5 second. Dont know if thats supposed to be so and dont know where to write u thats why i just used this Issues Tab to contact u about it. I noticed that when booting Kodi (libreelec) , Lakka , Linux like Ubuntu and Fedora. Even if i dont use Splashscreens its same with the stock Hekate IPL Bootlogo. 

Its nothing dramatic , just a cosmetic thing that doesnt look appetite :) maybe u can take a short look at it someday when ur bored . Thank you

Check the Video u can see the problem i have 👍 at the end of video

https://user-images.githubusercontent.com/96496833/221041658-12f56046-2c9e-465c-a84c-74f1b7a7cbcd.mp4

CTCaer, 2023-02-23:
That's the ram initial training.

So, I used to hide that with a backlight fade in/out before the L4T loader.
Thing is that I can only do that for Erista (2017) models.
On mariko (2019/lite/oled), I can't hide it.

So I took the decision to maintain **consistency** instead of **UX**, since UX could only be done for Erista models.
(I removed the backlight fade out/in basically)

Technical: Now why that happens?
During dram training, there are certain training hw functions that actually stall any access to RAM. When that specific training happens, the display hw logic memfetcher can't read the framebuffer's pixles at that point and just continues to try the next ones.
These also last way less than 1ms, reason why you see glitches and not the whole screen going to hell. And it happens as many times as how many freqs the table has.

TL;DR: It's here to stay

closed 2023-02-23 by CTCaer

lowburger, 2023-03-05:
@Bladen123 I may ask where can get your icons for hekate launch menu? Sry for off topic.

Bladen123, 2023-03-05:
I did them myself , its no problem that u ask im happy that u like them . I
can give u all Icons i did Just give me couple minutes to start my iMac and
compress them to a zip for u and i Forward them to U.

Bladen123, 2023-03-05:
Will Post them Here If CtCaer isnt against or against the GitHub Rules.

CTCaer, 2023-03-05:
It's not against my personal rules (as long as this offtopic discussion doesn't get prolonged)
or github ones (as long as they are not hundreds of MBs).

Bladen123, 2023-03-05:
ok its only 90 kilobytes zip. here u have it.
[Hekate Icons.zip](https://github.com/CTCaer/hekate/files/10892146/Hekate.Icons.zip)

lowburger, 2023-03-05:
@Bladen123 Big thx! Icon look really great. You have other graphic you made like bootlogo? Kodi bootlogo look very good!

That last question from me! For keeping issue clean we can delete those message, I now delete original message from me and will delete this message on next days.

Bladen123, 2023-03-05:
Just write me Here ***@***.*** , Sure i can do graphical Icons
and Bootlogos for u

lowburger, 2023-03-05:
@Bladen123 Github hide email address, so you must write in different way.

Pls delete our message in github issue because CTCaer would like not a off-topic discussion here :)

Bladen123, 2023-03-05:
Just Email me at ( Bladeneaera now e-mail Symbol than Gmail dot com 

I cant delete your Messages U have to delete yourself i think

lowburger, 2023-03-05:
@Bladen123 I have sent mail to you, remember check spam if not arrive :) Yes I delete my message. Thx.

Masamune3210, 2023-03-05:
Just a heads up, its probably not the smartest to send your email address to everybody who happens to be watching issues on this repo. While you can delete your comments afterwards on the site, the emails still get sent, meaning that they have a log of everything you have said