CTCaer/hekate #833: Nyx not working for hekate 6.0.2, HOS 1.6.0 on OLED Switch

MSco, 2023-02-24:
When booting into hekate, I get the message "Failed to init or mount SD! Press any key..."

When I press any key, I get to hekate 6.0.2, but without nyx gui interface. On the bottom it says: Nyx is Missing!
But I unzipped all the stuff of the hekate-nyx.zip (from hekate github) to the root of my sd card.

I am on

Switch OLED
OFW 16.0,
CFW 15.0.1,
Atmosphere 1.5.0 (pre-release),
hekate 6.0.2

I can boot to the firmwares by navigating with + and - and Power Button.

However, when my switch goes into standby, it doesnt awake anymore ... I always have to hard reset.

I removed atmosphere and bootloader folder, redownloaded atmosphere and hekate and unzipped compeletely fresh to my sd, without any custom config files with following versions of atm and hekate:

atm 1.4.1 + hekate 6.0.1: Works fine (But, of course, I cannot boot into ofw, since 1.6.0 is not supported by hekate 6.0.1)
atm 1.4.1 + hekate 6.0.2: Not working
atm 1.5.0 + hekate 6.0.2: Not working

I get this error, since I upgraded my ofw to 1.6.0. When I had OFW 1.5.1 and hekate 6.0.2 and atm 1.4.1, everything was working fine. But then I accidentaly updated my ofw and hekate does not work anymore.
For me it seems, that hekate 6.0.2 is not working for 1.6.0 (maybe this is only for OLED/mariko the case?).

CTCaer, 2023-02-24:
HOS has nothing to do with it.
You either have an issue with SD card (corruption or fake) or SD reader. Which is a strong indication that the issue happens in hekate and not nyx. Atmosphere is also irrelevant with that.

So, for starters:
1. How you paste hekate? Do you safely unmount the sd afterwards?
2. Get a pic from console info -> sd card. One from 6.0.2 TUI, since you can get into Nyx. And one from Nyx with 6.0.1.

FantasyDH, 2023-02-25:
I have the same problem.
but only on sandisk a2(512G),
sandisk a1(256G);samsung red(128G);samsung white(128G) no fault

https://img1.imgtp.com/2023/02/25/lBVgjSoP.jpg
https://img1.imgtp.com/2023/02/25/h9QL34g7.jpg

PS.
"When I press any key, I get to hekate 6.0.2, but without nyx gui interface. On the bottom it says: Nyx is Missing!"
select <reload> can enter the nyx gui interface normally,but reboot will report error again  "Failed to init or mount SD! Press any key..."
https://img1.imgtp.com/2023/02/25/1325RuOT.jpg

MSco, 2023-02-25:
That's interesting. I am on a 512gb Sandisk, too.

To CTaers question 1: I pasted the freshly downladed and unzipped hekate and atmosphere folders to a an sd, where i removed the old atmosphere and bootloader folders before. And I always safely removed the sd card after unmounting. It is interesting, that this works fine for hekate 6.0.1, but not for 6.0.2.

MSco, 2023-02-25:
[image]
[image]

CTCaer, 2023-02-25:
6.0.2 was working before updating HOS?
Also I don't get how you get failed to init sd but have 0 errors showing in sd card info in TUI.
Anyway, pm me in discord. You can find my tag via retronx discord.

Ozfer, 2023-02-27:
I am having the same issue on my OLED switch with 1TB FAT32 SanDisk Ultra.  1.4.1 + hekate 6.0.1 works but not Hekate 6.0.2. Didn't work for me even before updating HOS.

Ozfer, 2023-02-28:
I got it working using atm 1.6.0 + hekate 6.0.2 + NYX.bin 1.4.0

li1345825138, 2023-03-02:
I just update my switch OFW to 16.0.0, and use hekate 6.0.2.
When I reboot into OFW, it give me blue screen.
[image]
I didn't update my CFW version (still 15.0.1) and it work fine with hekate 6.0.2.

MSco, 2023-03-02:
The problem is solved for me. I had to update my sdloader to get it working: I have a switch oled modchip, that is non-flashable. After downloading the sdloader update from hwfly-toolbox and updating only the sdloader (not the firmware!), it was working fine.

closed 2023-03-02 by MSco

Excitedguy, 2023-03-31:
> The problem is solved for me. I had to update my sdloader to get it working: I have a switch oled modchip, that is non-flashable. After downloading the sdloader update from hwfly-toolbox and updating only the sdloader (not the firmware!), it was working fine.

Dude thank you so much! This is what I had to do on mine as well.

dnlflores, 2024-05-04:
Hi I am also having this issue,  I tried updating my sdloader but I am still receiving an error saying that it cannot init the SD card even though I can see SD card stats just like in the screenshot posted earlier.