CTCaer/hekate #834: can you add an option to delete the custom themes folder?

MichaelRebirth, 2023-02-25:
hi, small request for annoying problem...

can you add an option ini hekate to delete the custom themes folder atmosphere/contents/"01000000001000"

so if the user accidentally apply incompatible themes and they end up stuck in error blue screen. they dont have to pull the memoricard out ns and delete manually that folder using computer... and just can use hekate to fix it.

thanks

urherenow, 2023-02-25:
You haven’t had to remove the card in ages, because of Hekate. It has a usb function

MichaelRebirth, 2023-02-26:
still you need a computer just do delete that 1 folder.... the point is to be able to delete that folder without computer.

CTCaer, 2023-02-26:
No such cluttering feature will be added, just to save 30 seconds every 2 to 6 months.

closed 2023-02-26 by CTCaer

impeeza, 2023-03-01:
You can use another payload chainloaded from Hekate to that, by example tegraexplorer