CTCaer/hekate #836: Blank emummc.ini file when change emuMMC to SD File Based on hekate

apolope, 2023-03-06:
When I created a emuMMC to SD File, all process occurred ok, but when a change the emuMMC to this SD File, the hekate v6.0.2 create a emummc.ini file without any character legible, just a file with lots of whitespace. And switch os with emunand it does not start.

urherenow, 2023-03-06:
You didn't backup your SD before doing such a thing? The process does try to backup and restore things when partitioning and such, but that is obviously limited by available ram. 

I had to manually backup and convert my SX raw partition emummc last night, because nyx couldn't find boot0/boot1 for some reason. After backing up emummc on my PC, I placed a fresh Atmosphere+Hekate on the card and used nyx to repartition/format, then started to copy my things back. The repartition/reformat had no problem copying Atmosphere, Hekate, and a couple of sysmodules into ram and placing them back afterwards.

At any rate, the front page tells you every option to make a new .ini, and there are plenty of examples on gbatemp.

CTCaer, 2023-03-08:
Your sd card is fake.

closed 2023-03-08 by CTCaer

apolope, 2023-03-08:
> Your sd card is fake.

Exactly, after testing with H2testw (https://h2testw.en.lo4d.com/windows) I identified the problem, changed the SD and "voilà".

Thanks!