CTCaer/hekate #838: Can’t boot up

TrueTexan21, 2023-03-10:
I was playing ACNH and needed to reset my modem and router. Once it was connected again I tried accessing the internet in ACNH and was hit with an error telling me that I needed to update the system. 

Someone in Hangouts suggested that I update since the current release is stable enough. 

I updated Hekate and Atmosphere and I get this error as soon as I try to load up CFW. Any suggestions on how I can fix this?
[image]

Masamune3210, 2023-03-10:
1. This has nothing to do with hekate
2. Delete your custom home menu theme

TrueTexan21, 2023-03-10:
Oh, okay. Thanks so much. I didn’t realize it would cause an issue.

Masamune3210, 2023-03-10:
Custom themes are irrevocably linked to the version of Home Menu that they were made with, when you update the firmware you must remove them or else this will happen every time

TrueTexan21, 2023-03-10:
I’ve removed the whole “downloads” folder that contains the custom themes
but I’m still getting an error. Is there something else I must do in
addition to removing the files?

Masamune3210, 2023-03-10:
You must remove the actual theme and not just the downloads. Look for a folder with the same name as the title ID in atmosphere\contents

urherenow, 2023-03-10:
The error tells you which folder to delete. 0100000000001000. How do you think we know it’s your theme causing this? Try googling for 2 seconds before making an issue. Perhaps even look at the closed issues. This only happens to people like every single major firmware update…

Masamune3210, 2023-03-10:
Please, for the love of God, chill out on being so malicious.

urherenow, 2023-03-10:
> Please, for the love of God, chill out on being so malicious.

For the love of god, stop entertaining people posting on github issues, when it isn’t an issue with the project in question. Especially when it’s been asked and answered several times, and your “help” is cryptic as hell. You could have mentioned that the program ID in the crash screen, is the name of the contents folder that should be backed up then deleted. Then Google that same number if it doesn’t fix the issue (which would also lead to checking if another folder is present.

Github isn’t a support forum.

closed 2023-03-11 by CTCaer