CTCaer/hekate #840: Possible issue with digitizer init on (faulty?) Switch Lite

VioSynthax, 2023-03-15:
I have a Lite console where touch isn't functional in Nyx, but works normally in Horizon OS.  "Calibrate Touchscreen" fails, and "HW & Fuses Info" shows "Failed to get touch info!"

I can't confidently say that the board is not faulty, but whatever may be wrong with it isn't affecting HOS, but does prevent touch from being used in Hekate. This is a "fresh install" of Horizon I suppose you could say, as the console needed some board level repair and booted to a purple screen, so eMMC image was built from scratch with PRODINFO and keys from Lockpick_RCM.

CTCaer, 2023-03-15:
You can paint all over the screen in touchscreen test in HOS?

VioSynthax, 2023-03-15:
Yes indeed I can

CTCaer, 2023-03-15:
That's weird. No comms in Nyx is not normal.
Pm me in discord. You can find my tag via retronx discord.

CTCaer, 2023-03-15:
The verdict is hw issue, since HOS also exhibits that issue from time to time.
(Nyx only tries to init 3 times and then stops interfacing with the touch IC completely.)
And all voltages are there (i2c 1V8, vddio 1V8, 2V9 vdd and rst pin at logic high).

closed 2023-03-15 by CTCaer