CTCaer/hekate #842: Error code LR 400E720

Synxian, 2023-04-01:
While using the usb mass sotrage tool to change some files in the atmosfere folder, my file explorer stopped working and the switch screen started to have some interference (the same as the photos), then i started to get this error:
`hekate exception occured (LR 4000E720):
Hang detected in LP0/Minerva!`
I formated my sd card, but the error persisted, and sometimes this happens to the screen:
[image]
[image]
[image]
When loading whit the OFW, the screen goues black after nintendo logo

- Sd card formated with fat32

balika011, 2023-04-01:
Memory issue. Probably your switch was dropped and it has a bend in the case looking from the charge port side.

Synxian, 2023-04-01:
I don't see any bends,  and i was playing perfectly fine yesterday, on another hand, now i turned it on with the ofc and the screen was blue

Synxian, 2023-04-01:
> Memory issue. Probably your switch was dropped and it has a bend in the case looking from the charge port side. The most common fix is reflow the cpu.

As i don't have the tools nedeed for a reflow, i'll have to wait till monday, thanks for the reply, after the blue screen i think it's probably this

SocialleyWebsite, 2023-04-05:
Same issue with mine

closed 2023-04-06 by CTCaer