CTCaer/hekate #843: White screen on switch OLED

zakwarrior, 2023-04-08:
Hello,

I'm getting a 80/90% white screen when i load now hekate on switch oled, it's not a full white screen, i cannot upload a picture since my phone broke few days ago and i'm waiting for a new one, screenshot fonction doesn't work i tried to do with this white screen but it doesn't work, so i uploaded an image that i did with pain to show you what it looks like

This happens just after hekate logo, so the start logo show up normaly then when the time bar is finished it get that
I when back to 6.0.2 and everything works fine like it should
[image]

CTCaer, 2023-04-08:
Did you redownload the release and repasted it and safely eject to make sure it wasn't corrupted?

zakwarrior, 2023-04-08:
Yeah I'm well aware for that since I also code programs and scripts, I've been doing this many times the same way, i will try again tomorrow but i always prefer to delete all the older stuff on the sd and replace with new files since I have my ini files and the rest saved. Never had this issue before

Envoyé à partir de Outlook pour Android<https://aka.ms/AAb9ysg>

closed 2023-05-07 by CTCaer