CTCaer/hekate #845: Optimize relocator function in bootloader entry

paulober, 2023-04-16:
### Description 
The current relocator function in `start.S` uses `LDMIA` and `STMIA` instructions for loading and storing multiple registers while copying the bootloader code to the correct location in memory.

A possible improvement to this function could be using `LDRD` and `STRD` instructions for performing doubleword (8-byte) transfers, which might improve efficiency. This optimization assumes that the source and destination memory regions are 8-byte aligned.

### Proposed change:

Replace the existing `_reloc_ipl` function with the following optimized version:
```asm
_reloc_ipl:
    LDRD R4, R5, [R0], #8
    STRD R4, R5, [R1], #8
    LDRD R6, R7, [R0], #8
    STRD R6, R7, [R1], #8
    SUBS R2, #0x10
    BNE _reloc_ipl
    /* Jump to the relocated entry. */
    BX R3
```

paulober, 2023-04-16:
I would implement this and PR if someone with more advanced knowledge of this repo/project reviews this change.

CTCaer, 2023-04-16:
LDRD/STRD are not supported on arm7tdmi (armv4t). Armv5te added these if I remember.
Easy to see during compilation though. Compiler should error out if not supported.

CTCaer, 2023-04-16:
Yep just checked the reference manual.
They are not supported.
Thanks anyway

closed 2023-04-16 by CTCaer