CTCaer/hekate #846: FatFS error code 5

mvfsullivan, 2023-04-17:
I recently grabbed my old Switch and wanted to update FW from 14 to 16 and AMS to 1.5.1, just update everything. Anyway long story short, I ended up deleting my SD card to "start fresh" because AMS was failing to load (that typical Atmosphere error screen, something about 2001-0123. I got a pop up on Windows about deleting "System Volume" folder and I noobishly deleted it. I think thats the root issue here.

I have formatted in Fat32, exFat, NTFS. I cannot boot into Hekate or anything. I plug in my rcm payload, click the power and rcm does its thing, then I see the FatFS error code 05. 

I cant boot into the stock OS because I have autorcm on but I cant even turn that off now. Please send help!

CTCaer, 2023-04-17:
Don't follow random guides.
Also system volume is fine to delete, but you probably somehow corrupted some fat records.
The error means that file or folder not found. And hekate on boot only checks stuff in /bootloader.
(Although, all sd cards having the same issue is weird.)

You can try to delete bootloader folder, after copying your configs ofc (hekate_ipl.ini and ini folder), and then repasting release.
That's the only method to fix corrupt directories without formatting btw. Only repasting it, doesn't work.

If that doesn't work then:
If you have data in it, back it up on pc, since following the steps below will delete everything.

Then format it with https://www.sdcard.org/downloads/formatter/

After that paste hekate in it and launch it. Then format using Tools -> Partition SD card.
Make sure to select every partition you want before continuing, since if you run that again, you will wipe your card again.
When that's done, copy back everything that's needed.

mvfsullivan, 2023-04-17:
Its recommended to format to FAT32 right? Why doesnt that sd card formatter tool not give me the option?

mvfsullivan, 2023-04-17:
Ah that program in the guide I followed I think screwed up my card. I can see it in device manager, but it doesnt show in File Explorer. SD Card Formatter sees it as \\volume with a bunch of letters. I can "format" it but it doesnt come up in Windows anymore

urherenow, 2023-04-17:
Chances are, your card either died, or it's fake. If it's real and a SanDisk, you can get a replacement for free.

mvfsullivan, 2023-04-17:
I've been using this 400GB card for years and have had it nearly full. I also tried a 32GB card and the old 128GB card I had in the Switch before I upgraded.

Also all cards are recognized by my computer.

I also just tried to double check the SD reader motherboard and its definitely connected. The Switch does "see" the card too because if I try to boot without it, it says to insert a card. I am still stuck with this issue.

> Don't follow random guides. Also system volume is fine to delete, but you probably somehow corrupted some fat records. The error means that file or folder not found. And hekate on boot only checks [...]

I tried all of these steps and no luck still!

urherenow, 2023-04-17:
AutoRCM… so this is a v1 erista? Inject hekate with your PC put your card back in afterwards

CTCaer, 2023-04-17:
> Its recommended to format to FAT32 right? Why doesnt that sd card formatter tool not give me the option?

Completely irrelevant. hekate's partitioner will format to fat32 anyway. So exfat and fat32 are both fine if you are gonna use the sd partition tool. And even if you don't have emummc or linux or android, you should still use it so it can be formatted in fat32 with performance optimizations.

> I tried all of these steps and no luck still!

Well can't help you, since that issue exists on your side.
All I can say, is to make sure that you redownloaded release, as to make sure your previous copy was not corrupted or partial.
And that you are safely ejecting your sd card from pc.
Plus you never said where that happens and hekate/Nyx do not have the exact `FatFS error code #` error message anywhere, and it makes me wonder why you created an issue here, so..

mvfsullivan, 2023-04-17:
> AutoRCM… so this is a v1 erista? Inject hekate with your PC put your card back in afterwards

Oh my god why did I not think of this. Immediately worked.

Now onto the actual issue I was originally facing, seems if I download all the latest stuff, I cant get past Atmospheres crash screen with error "20013 0123" or whichever, something like that. Theres no other info

mvfsullivan, 2023-04-17:
The FatFS error code 5 issue was literally my fault. I have a Dragon Injector and thought the folder with Fusee was supposed to be named "dragoninjector" but its actually "dragonboot". 

Still facing the Atmosphere issue with 16.0.0 though!

CTCaer, 2023-04-17:
You are in the wrong place to ask support for that.

mvfsullivan, 2023-04-17:
A few reboots later and it magically worked. All good