CTCaer/hekate #849: ito test : force open (1) chn: 2 the touchscreen calibration failed

Chekeleke, 2023-04-27:
I Replaced the Digitizer. the touchscreen works but some spots don't. When I try to do the screen calibration I get this error ito test : force open (1) chn: 2 the touchscreen calibration failed. Tried my times not lock. the digitizer is fine because I installed android and run screen touch pass all the spots were fine. Any solution pls? thanks

Chekeleke, 2023-04-27:
Here are some pictures. The first  one is the error and the second one the test I did on android 

[image]
[image]

Chekeleke, 2023-04-27:
I left the dot floating  on purpuse, if not will close the windows  as finished. Here is a picture   of HW info 
[image]

CTCaer, 2023-04-27:
Panel is paired to the touch ic, so it's a connector/flex/hw issue.

closed 2023-04-27 by CTCaer