CTCaer/hekate #850: Failed to apply ‘emuMMC’. Failed to launch HOS.

troubleshooo, 2023-04-29:
I recently updated hekate and atmosphere and now when I try to boot in to my emuMMC I receive the follow error:

Initializing...
Found pkg1 (‘20220801142548’)
Identified Pkg1 and MKey 14

Atmosphere 1.5.2-434C8CEF via FSS0/PKG3
Max HOS: 16.0.1
Unpacking … done!
Loaded config and pkg1
CFW mode
Generated keys
Loaded warmboot and secmon 
Read pkg2
Parsed ini1
Patching kips
Decomping ‘loader’, sect0, size 106732..
Applying ‘nosigcheck’ on Loader, sect 0
Failed to apply ‘emuMMC’!
Failed to launch HOS!

Press any key 

Any help would be appreciated.

Leerz, 2023-05-02:
I would check your sdcard first if it's authentic. rune fakeflashtest on it. also check if emmc is not in slow mode

CTCaer, 2023-05-07:
You are probably missing files from sd or you have corruption.
In the latter case you might need to delete bootloader folder.

closed 2023-05-07 by CTCaer