CTCaer/hekate #853: [Enhancement] KIP patches for faulty power gauge ic

josete2k, 2023-05-07:
I have an oled with (maybe) faulty power gauge ic (max17050)...

As hekate boots but the power batt is shown as 0v (temp is 0 too) but HOS doesn't boot anyway, should be posible to bypass this issue in order to boot HOS?

CTCaer, 2023-05-07:
These would never be provided by hekate.

Anyway, you can patch out the checks in boot kip (stock or atmo one). You will still need to patch ptm sysmodule though.
Or, just simply fix your fuel gauge.

Mind you that it's bad to use it without one, since battery charging can't be finely controlled by OS.

closed 2023-05-07 by CTCaer

josete2k, 2023-05-07:
Ok, Thanks...

I'm not 100% sure if that is a faulty gauge ic, so I thought that booting with this check removed should be a good way to test.

I'll take a look, Thanks.