CTCaer/hekate #854: Multiple emunand not possible/not working?

kapeas, 2023-05-08:
Hello, recently decided to update old switch, and installed new bigger SD.

Problem description:
Can't make my hekate_ipl.ini file boot multiple emunands (2 RAW and 1 SD). Works perfectly if I use the change emuemmc. If I don't, all my entries goes to the emunand selected through the change emummc option, ignoring the rest of the file and all different emummc paths. All 3 emunands have different "console nicks" so I can tell exactly wich one is booting. Even if I press RAW1 and the one loaded is the one selected through change emummc.

I think I may be missing emunand.ini but I can't seem to find valid examples. I think I must edit RAW partitions sectors there, or maybe that is from older versions. Anyway, some help/pointers/advice would be greatly appreciated.

I tried to use a multi emunand setup, but maybe this is not possible or I missedunderstood some of the isntructions.
Used some templates found online since I wasn't able to fix it on my own, so, it is possible that introduced unnecessary/wrong lines or key/values. 

I have 2 RAW and 1 File Based emummcs.

My hekate_ipl.ini looks like this (commented some lines since could't make it work with the 3 of them, but you will get the idea.)

[config]
autoboot=1
autoboot_list=0
bootwait=3
autohosoff=0
autonogc=0
updater2p=1
backlight=100

[CFW - EMUNAND ¿RAW1?]
fss0=atmosphere/package3
kip1patch=nosigchk
atmosphere=1
icon=bootloader/res/iconopestoso1.bmp
payload=bootloader/payloads/fusee.bin
emupath=emuMMC/RAW1

[CFW - ¡¡¡¡¡sysMMC!!!!!]
fss0=atmosphere/package3
kip1patch=nosigchk
atmosphere=1
emummc_force_disable=1
icon=bootloader/res/iconopestoso2.bmp

[¡¡¡¡¡STOCKSYS!!!!!!- sysMMC]
fss0=atmosphere/package3
emummc_force_disable=1
stock=1
icon=bootloader/res/iconopestoso2.bmp

[Lockpick RCM]
payload=bootloader/payloads/Lockpick_RCM.bin
icon=bootloader/res/iconopestoso1.bmp

[Otro PAYLOAD RCM]
payload=bootloader/payloads/Lockpick_RCM.bin
icon=bootloader/res/iconopestoso1.bmp

#[CFW - EMUNAND RAW2]
#fss0=atmosphere/package3
#kip1patch=nosigchk
#atmosphere=1
#icon=bootloader/res/iconopestoso1.bmp
#payload=bootloader/payloads/fusee.bin
#emupath=emuMMC/RAW2

#[CFW - EMUNAND FILEBASED]
#fss0=atmosphere/package3
#kip1patch=nosigchk
#atmosphere=1
#icon=bootloader/res/iconopestoso1.bmp
#payload=bootloader/payloads/fusee.bin
#emupath=emuMMC/SD00

Thanks in advance for the help. Sorry if this is already solved, didn't find it on recent issues.

Regards,
Kapeas.

kapeas, 2023-05-08:
Using Hekate Version 6.0.3 with nyx

urherenow, 2023-05-08:
Fusee.bin doesn’t know a thing about multiple emummcs (to my knowledge). It only knows if emummc is enabled, by checking emummc.ini (which iirc is in the root emummc folder) Try with fss0 and package3.

closed 2023-05-09 by CTCaer

kapeas, 2023-05-10:
Ok, I guess I missunderstood the concepts when saw the multiple nands in the template.
Thanks for the replies and help. 
Closing the issue.
Regards,
Kapeas.

urherenow, 2023-05-11:
What I think you misunderstand is that fusee is its own bootloader and doesn’t look at hekate’s .ini file. If you want to be sure you’re using hekate’s features, don’t use payload=fusee.bin.

impeeza, 2023-05-12:
This is not a support site, please redirect your question to a proper space like discord or GBATemp i do recommend read the guide https://rentry.org/switchHackingIsEasy and the README.MD of this repo

CTCaer, 2023-05-12:
@impeeza don't post piracy guides ever again
plus that it uses hekate completely wrong and it's counter intuitive, especially with what user asked.
Consider that your last warning.