CTCaer/hekate #859: Backup eMMC error

LePumpkinKing31, 2023-05-20:
I really hate having to ask, but during the eMMC RAW GPP backup it hits 95% at rawnand.bin.13, does 3 failed passes, then aborts.

CTCaer, 2023-05-22:
Get screenshot of emmc info pop up and below pop up and also sd card info screenshot.
All that **after** a failed attempt to backup emmc.

CTCaer, 2023-05-26:
You didn't send the most important thing I asked.

Get screenshot of **emmc info** pop up and below pop up.
All that after a failed attempt to backup emmc.

That means go to console info -> emmc info and get a screenshot of the pop up that will show up in there. Then close it and get a screenshot of the actual emmc info.

LePumpkinKing31, 2023-05-27:
Didn't realize what you meant rip. It fails pretty consistently and only takes a few minutes so its ok. Hopefully this is what you needed.

Before fail (actual emmc info)
[image]

After fail
[image]

CTCaer, 2023-06-09:
Have you ever tried to continue the backup from where it was left?
It clearly fails on part 13, so you can just add the index file back and it will continue from there.
Here, paste the `partial.idx` from that zip into sd root and it will continue from part 12.
[partial.zip](https://github.com/CTCaer/hekate/files/11708671/partial.zip)

Anyway, that's a hw issue and I can't help fixing it.

closed 2023-06-09 by CTCaer