CTCaer/hekate #862: Issues while loading Partition based emuMMC, currently remaking the emuMMC in the same partition

BeboBabaYellow, 2023-06-05:
what files have to be in the emuMMC for it to load? all guides i have followed have had different issues and my current issue is 
"Unknown pkg1 Version. Unable to launch HOS!"
it uses a backup loader once the first one fails, but the backup also has the exact same error

when attempting to load into fusee.bin as a different guide suggested i just got the error 
"Invalid GPT Signature"
not really sure what it means but i dont plan on using that method ever again, but it may be useful to you
Do i have to copy all files from the root of the sd card to the root of the partition? (assuming it would be its own root as its separated from the rest of the sd card)
my nintendo folder is bigger than 130 GBs so i REALLY hope i dont have to copy the whole thing over because that will likely take a whole day

BeboBabaYellow, 2023-06-05:
iam solving my own issues here, apparently im not supposed to access that drive, and iam currently installing tinfoil on a supposedly working modded switch, im just here to ask is there a guide to make sure its safe to play on the stock switch? as i have atmosphere and a few more files on the root of the sd card which the switch can read, is it safe to boot into stock and play online on something like splatoon 3? and on emunand would it also be possible to play online?, after answering this question you can close this issue if you want, id prefer if you didnt and let me do it myself but go ahead

BeboBabaYellow, 2023-06-05:
album isnt detecting switch file with all apps inside, hoping to find my own solution but if you are reading this and you know of any, comment a list and ill try em out

impeeza, 2023-06-05:
As you need support and are not reporting a real issue or code bug, is better you go to a support site like discord or GBATemp, there you will find help.

CTCaer, 2023-06-05:
You are corrupting things as it seems. Otherwise your card is fake.
Also piracy not supported.
Plus do not spam.

closed 2023-06-05 by CTCaer