CTCaer/hekate #863: Nyx exception occurred <LR 81064C78>

Bonanno1234, 2023-06-10:
I cannot press any button, and need to restard alot of time to make this work again, is these snyway to fix this thing? Thank you
[image]

CTCaer, 2023-06-10:
You are not fully up to date.
Update to latest first and then when it happens post that message again so I can see where it dies (the number 81064C78 will change).

Bonanno1234, 2023-06-10:
> You are not fully up to date. Update to latest first and then when it happens post that message again so I can see where it dies (the number 81064C78 will change).

Thank you but i have a quick fix by use TegraRCMGui and choose Fusee to boot on EMMU. My brother has take the switch and go to travel :(

CTCaer, 2023-06-10:
Well anyway, if it happens again, it's either corruption of the bootloader folder and files or you must use `bpmpclock=2` or `bpmpclock=3` in `nyx.ini`

closed 2023-06-10 by CTCaer

Bonanno1234, 2023-06-10:
Thank you for help me ^^ Have a good day <3