CTCaer/hekate #865: [feature request] dump Bluetooth pairing keys for audio devices from HOS to import them to andorid/Linux

zekromisblack, 2023-06-11:
This would be a incredible feature for people like me who has earbuds and don't wanna have to repair them every time I switch my OS

impeeza, 2023-06-11:
the bluetooth database do not include them?

OrGoN3, 2023-06-11:
You can dump the hos title, https://gbatemp.net/threads/pairing-the-joy-con-on-emunand-and-sysnand.562519/, not sure if that's enough.

impeeza, 2023-06-12:
Wait one moment.

Since version 1.5.1, Atmosphère have a new setting `enable_external_bluetooth_db = u8!0x1` on `SD:/atmosphere/system_settings.ini` file:

Once you enable that setting the first instance of atmosphère you run will copy the actual Bluetooth sync database to the file `SD:/atmosphere/bluetooth_devices.db` then on all instances of Atmosphère you run and if you set the value en on System_Setting.ini file will use the database on the dump file instead of the one on EmuMMC or SysMMC so, all your instances will share the same bluetooth database, including headsets and controllers.

And if you use MissionControl you can even spoof any mac from controllers synced with other consoles or PCs and you don't need to lose your sync with external devices.

More information on:

https://github.com/Atmosphere-NX/Atmosphere/releases/tag/1.5.1

zekromisblack, 2023-06-15:
> Wait one moment.
>
> Since version 1.5.1, Atmosphère have a new setting `enable_external_bluetooth_db = u8!0x1` on `SD:/atmosphere/system_settings.ini` file:
>
> Once you enable that setting the first [...]

that does not allow me to get the pairing keys from that database file, and i have no idea how its formatted in a hex editor

impeeza, 2023-06-15:
its a dump of the HOS database, so you can look on Switchbrew page about that information.

TechniXC, 2024-09-18:
@zekromisblack Hey! I think I found the sollution! It works for me to transfer non OEM joy-cons LTK to L4T, but you could do the same for all your BT devices.

You could open bluetooth_devices.db with https://hexed.it/ and on the first BT device mac start at 0x00000008.
Then LTK is the sequence of 16 bytes between 25:08 and 01 bytes. Aprox address of it's first byte is 0x00000031. Important thing that LTK bytes here in REVERSE order. That's mean if you have 78A62351XXXXXXXXXXXXXXD37C24 sequence in bluetooth_devices.db you should copy LTK as 247CD3XXXXXXXXXXXXXX5123A678.

organizedkaos, 2025-01-10:
I just recently discovered and installed the picofly chip and hekate etc. etc.  I have a good grasp of how everything goes together considering I have been at this for 2 weeks now.  My 3rd party (spican jc200) bluetooth switch controller works between ofw and cfw emummc using the bluetooth_devices.db setting.  @TechniXC instructions allowed me to get the ltk for this controller but I can't seem to figure out how to make this information/controller work in android/linux without re-pairing.  Unfortunately, the controller only seems to have 1 pair memory so everytime I pair it to android/linux, I have to re-pair it to ofw and go through the bluetooth sync between ofw and cfw.  Nyx joycon dump only works for genuine joycons.  Is there any way the nyx joycon dump can read the bluetooth_devices.db and create a similar file (joycon_mac.ini/bin) or would the android/linux builders have to create the feature (reading the bluetooth_devices.db file) on their end?

organizedkaos, 2025-01-13:
I'm not a coder but I can write some bash and I can understand some code.  But again, I am not a coder.  I managed to write a bash script that grabs the information needed for the joycon_mac.ini file (device mac, system mac, ltk) from the bluetooth_devices.db file and the switch.cal file and generate a joycon_mac.ini file with the properly formatted ltk.

I am doing this for my 3rd party wireless bt "pro controller" but ubuntu is still not working to accept the joycon_mac.ini information.

After I did this, I was wondering if an option could be added next to the "dump joycon bt" button to "dump hos bt pairs" to the same file joycon_mac.ini by reading the file that contains the bt information from hos?  I used the "syncfix" script to make the hos bt pairs work in atmosphere and apparently it (syncfix script) copies a "save" file name 8000000000000050 from sysnand into emummc and thats where the bt pair information is stored.

Alexgamer470, 2025-04-14:
any updates on this? i'd need this feature too, or at least a proper workaround since it really is a pain to always repair when i switch to linux/hos (@TechniXC advice seems good but i cant quite follow what he's saying).

CTCaer, 2025-04-29:
Since that database is not synced back to emmc, it's not really a fully viable solution, but a "with conditions" one.
So it doesn't fix the issue of OFW/Stock booting.

nyanpasu64, 2026-11-16:
I've written up a script and tossed it at https://codeberg.org/nyanpasu64/dump-bt. It seems that `name` goes unused and `name2` is instead being populated.