CTCaer/hekate #866: "V1.5" Nintendo Switch RCM black screen after injection

XalyLodak, 2023-06-13:
Hi,

My V1.5 Switch is RCM-ok but the injection stuck on the black screen, do you have a PayLoad to force de re-injection and the restart of the LCD "service", explain me about the TegraRCM language or engineering, I have weeks to kill (3 month of vacation so I want to work on that)

More info :
CPU : Intel Core i7 7700HQ
GPU : GTX 1050Ti
OS : Ubuntu 23.04 Linux 6.1
Switch : 16.0.1OFW TegraRCM-ok but RCM semi-patched with black screen

I work in contribution for the Linux Kernel

masagrator, 2023-06-13:
All Switches are "RCM-ok", it doesn't mean though that every unit is exploitable. If TegraRCMGui returned stack size 0x0000, it is patched and you cannot use this method to hack Switch.

Masamune3210, 2023-06-13:
All "RCM-ok" means is that you successfully entered RCM mode. Its important to remember that RCM isn't anything exploit-related directly, its a recovery mode present in almost all Tegra systems. The exploit only exists in older systems that haven't been patched to fix nvidia's bug.

impeeza, 2023-06-13:
what is a 1.5V switch console? if is a console manufactured after July of 2018 it's for sure a "patched" console therefore you need a physical method to run Hekate on it.

closed 2023-06-14 by CTCaer