CTCaer/hekate #871: [Feature Request] Remove 4GB to 12GB emuMMC resize limit

nganjehloo, 2023-07-10:
I was wondering if it is possible to allow for a partition of any size up to 29GB FULL (not 4GB-12GB, 29FULL +)? If not I am curious to know why this limitation exists.

CTCaer, 2023-07-10:
Because there's no point. It's resized so it can be small. Not just an arbitrary size.
The data in it also needs to be formatted, so that contributes to reasons even more.
And if, at some point, I add backup/restore and USER data fits in it, if it was unconstrainted it would take forever.
Another one is that game data can just go to sd fat. And there's no need for >12GB for save data (afaik).
Lastly, it complicates the underlaying code too much for such an edge case.

Clydefrosch22, 2024-03-05:
Not to necro this issue, but I came past this since I'm also trying to set up emummc of more than 12gb because I also think 50+ on an oled is too much, but 12 seems just slightly too little.

minecraft-ish games come with rather unholy large save files, usually per user, so things ballon real quick on a shared system.
 Lego World has 2gb per user (which seems to be the general limit a save file is allowed to have because it's also the largest I've seen on  minecraft save yet), dragon quest builder taps out at just about 800 per profile, skyrim can reach like 300-400 too and there's more than enough small games that come with 128mb files.

Now granted, not everyone plays these, but it's not exactly impossible to fill 12gb up with somewhat popular games