CTCaer/hekate #872: Nyx missing :/

Moshi0ijbol, 2023-07-25:
Soo trying to mod my switch from android. but having major issues. Followed the switch homebrew guide. I have latest version of hekate. trying to launch to create a backup but i can’t launch! nyx is missing and “Failed to init SD Card.” error pops up. I feel like i’ve hit a wall and can’t find answers anywhere ‘-.- help please. I absolutely beg of you

impeeza, 2023-07-25:
Hello there, this is not a support site, This is a BUG REPORT zone, please read the documentation on https://github.com/CTCaer/hekate/ there is all you need.

May be your card is damaged or your console have a broken sd card reader.

for further support go to a right site, Discord, GBATemp are example of support sites.

Moshi0ijbol, 2023-07-25:
That’s the thing. The file is there. my sd card is new. got it on the 21st. my switch never had an sd card in it before so i highly doubt the reader is damaged

impeeza, 2023-07-26:
Anyway test it.

Moshi0ijbol, 2023-07-26:
i just did again.. still nothing😥

CTCaer, 2023-08-07:
You are supposed to extract the bootloader folder from the release zip in your sd card root

closed 2023-08-07 by CTCaer