CTCaer/hekate #877: Nyx exception occured ( LR 4003FEDC )

zebatator, 2023-08-16:
hello guy,

my switch does not boot and crashes an error "Nyx exception occured ( LR 4003FEDC ) UNDEF press any key".

sometimes the hekate menu opens on entering the date/time, but it's freezes before.

I tried:
-change SD card
-format SD card
-delete, recreate the directories
-finally factory reset the switch
nothing works

FW 16.0.3 / hk v6.0.5 / Nyx v1.5.4

can you help me ?
thank's a lot

CTCaer, 2023-08-17:
remove /bootloader/sys/nyx.bin
then boot it and go to console info and sd card and get a photo

zebatator, 2023-08-18:
hi, here is the photo
[image]
thanks for your help

CTCaer, 2023-08-18:
Card is legit so reextract the files from release to make sure there's no corruption.
If it continues having the same issue then edit /bootloader/nyx.ini (create it if it doesn't exist) and set `bpmpclock=2` or `bpmpclock=3` in `[config]` section.

example:
```ini
[config]
bpmpclock=3
```

zebatator, 2023-08-20:
hi,
```
[config]
bpmpclock=3
```
Now it's working !
The issue is solved

closed 2023-08-20 by zebatator