CTCaer/hekate #879: 2168-0002(0x4a8)

namujcd43, 2023-08-21:
I'm not really a home brewer by craft, but I downloaded the latest Hekate file I, booted well I, uninstalled all themes, and reinjected update payload into the switch. so it boots up fine and i can ply games but once i try to get into JKVS files  it gives me this code i tried to see if i had it in the atmosphere folder to delete in the contents but there is no folder there in the sd.  please help and if its not to hard can you break it down to me like a noob(i am) . thanks for the help.
[image]

CTCaer, 2023-08-21:
That's not a hekate issue. Ask JKVS dev.
But just in case run archive bit fixer in hekate -> tools -> 2nd tab at the bottom.
And make sure hbmenu exists.

closed 2023-08-21 by CTCaer

namujcd43, 2023-08-21:
Thank you for the  response  how would I be able to get the hbmenu folder if they're not there.