CTCaer/hekate #881: [DOUBT] Allow T210B01 RAM OC up to 3200 MHz (from 3000)

pedrozun, 2023-08-23:
Hi guys, with this new changes in Hekate v6.0.6 i read it:

```
Automatically set rated DRAM frequency for T210B01 by default.
Since Arachne is now final and stable use 1866 MHz for old ones and 2133 MHz for newer ones.
T210 will still use 1600 by default, since these chips are rated as such.
Allow T210B01 RAM OC up to 3200 MHz (from 3000) and T210 RAM OC up to 2366 MHz (from 2133)
Some ram models and binning can handle it. This needs L4T 5.1.2 release and up to work.
```

How can i achieve this OC using the switch?

CTCaer, 2023-08-23:
is that a doubt or question?
If it's a question, re-read what the text says.

closed 2023-08-23 by CTCaer

pedrozun, 2023-08-23:
> is that a doubt or question? If it's a question, re-read what the text says.

So the mariko chips automatically set the frequency to 2133mhz on newer models? and can reach 3200mhz automatically too?

CTCaer, 2023-08-23:
1. If you boot L4T on a T210B01 ram, it will be set to ram chips' rated clock, instead of 1600.
2. That's not what it says. It says allows. Previous max was 3000. And it's not automatic. User must manually edit the config to enable that.