CTCaer/hekate #882: black screen after launch

ventix31, 2023-08-23:
hello, sorry in advance for my english. I use a translator. 
                                                                                                                                                                                                                                                                                          i have a problem with hekate, when I launch the launch then atmo vanilla it gives me a black screen
I'm on the version hekate 6.0.6   my version switch is 16.1.0    
I already managed to get hekate to work at first.
I think the screen has been since the 16.1.0 update of the switch. 
do you have an idea ? thanks in advance

Lobeezy, 2023-08-24:
I had the same issue and found a work around. I to deleted everything in the bootloader and atmosphere files on the sd and reupload those file again. Hopefully this might help you.

ventix31, 2023-08-24:
thank you for answering me, I'll try it, I'll let you know if it worked

madmanmoe, 2023-08-25:
Hello. Having the same issue to. I've updated the switch to the latest update before trying as of 2023-08-24. I get to Hekate after injecting it from TegraRCM, but when i click on the launch icon, it goes black screen. Initially did all my SD formatting and backup using 6.0.5, then found that 6.0.6 was out and simply replaced all the bootloader files with the 606. Still having the same issue. 

I'm following the [Yuzu step-by-step guide](https://yuzu-emu.org/help/quickstart/) (except now using the latest hekate). I just reformatted my SD card again (this time to enable emuMMC partition and will follow the guide again. hopefully it will work out.

madmanmoe, 2023-08-25:
> Hello. Having the same issue to. I've updated the switch to the latest update before trying as of 2023-08-24. I get to Hekate after injecting it from TegraRCM, but when i click on the launch icon, [...]

re did everything from scratch again using 606 and now i don't get the black screen anymore. yay! now...according to Yuzu, it says hold the R button and click on any game to open Atmosphere...except that doesn't work. I looked online, and they said click on album and it will show...it did not. did something change?

urherenow, 2023-08-25:
It's news to me if hekate is supposed to work with yuzu. What are you trying to do with it? Hekate is a bootloader for the switch, and isn't needed in the slightest, for an emulator.

ventix31, 2023-08-25:
yes i don't understand either because my problem is on hekate

CTCaer, 2023-08-25:
This is for actual hekate issues not general hos support.
Also don't mix and match files from atmosphere or leave old incompatible sysmodules

closed 2023-08-25 by CTCaer